dayalannair/FMCW_FFT_Radar #3: FFT Improvements

dayalannair, 2022-06-11:
Hi @jpt13653903 
After adding Gaussian noise of up to 5 bits (noise between 0 and 32):
`n = 32*rand(200,1)';
I_row = round(I_row + n);
Q_row = round(Q_row + n);`

- The result did not change. I verified that the input data is as required. Not sure if adding more noise (i.e. increasing noise magnitude) will change FFT results. The rounding was needed to create fixed point input data.
- The floating point option in the FFT IP is for the case of floating point input data. I see that adding the noise to my real data converts the integer ADC value to a double precision value. I will try using this as the FFT input.

I will investigate these points further.

jpt13653903, 2022-06-11:
- 12 dB is 2 bits, not 5...  5 will work, but it's excessive.
- You're adding the "same" noise to I and Q.  They should actually be independent.
- The FFT must change (i.e. the noise floor should rise).  If it does not, you're doing something wrong.

dayalannair, 2022-06-11:
Ah okay. Will generate separate noise. IQ has a 90 deg phase shift, so should the noise have a 90 deg shift (not sure if that makes complete sense).
I.t.o bits, that makes sense as its a voltage, so 20log10(2^2) = 12dB. So for the max signal 20log10(2^12) = 72 dB the noise floor is around 60dB below. Also thinking now that there is no real headroom i.e. data goes up to the max value 4096 though the hardware etc was probably correctly designed --> out of scope of this project.

jpt13653903, 2022-06-11:
No -- I and Q really are independent.  If you use a Gaussian distribution for I and Q, the magnitude ends up with a Raleigh distribution, and the phase with a uniform distribution.  I'll leave it as an exercise to figure out why that is... ;-)

See [here](https://en.wikipedia.org/wiki/Quantization_(signal_processing)#Quantization_noise_model) for a summary on signal-to-quantisation-noise on a sinusoidal signal.

dayalannair, 2022-06-14:
I have managed to boil it down to this: the FFT possibly outputs a floating point value as stated below:
[image]
I am using the unscaled output, so the number may be full-precision (first point). I think this means single precision, Another option is that the data format parameter refers to both input and output data (so I could just have the input as .0000). 

However, the output is of size 24 bits per Im and Re, so it cannot be 32 bit precision. A final concern is the 'Phase Factor' parameter which I cannot find much info on. I have ignored it so far since here we are concerned with magnitude.

jpt13653903, 2022-06-14:
Phase is important if you want to follow the FFT with coherent integration...  By that is a topic for another day.

I would stick to fixed-point.  The price you pay in performance and area (for floating point) is generally not worth the effort.

dayalannair, 2022-06-15:
### Set input size to 16 bits, with the 4 MSBs set to 0
This was done as it seems the core is interpreting the unsigned input as 2's complement.
If the data is plotted as signed decimal, one can see the larger magnitude values wrapping around:
[image]

The plot below shows that the real FFT is correct. This is the case for inputs of 12 bit increased to 16 bit without sign extension since the values are unsigned:
[image]
For reference, when keeping the input as the original 12 bits, both the real and imaginary parts are different:
[image]

dayalannair, 2022-06-15:
ISSUE RESOLVED: extended 12 bit to 16 with zeros in MSB. This prevents unsigned being used as if 2s complement

closed 2022-06-15 by dayalannair