dayalannair/FMCW_FFT_Radar #4: Noise in real data

dayalannair, 2022-06-14:
Hi @jpt13653903 
If I am feeding real (as in actual radar data, not Re and Im) data to the FFT (which is 12 bits, unsigned) would that not already contain thermal noise which is 12 dB above the quantisation noise? Surely the front end sampling system would account for quantisation noise... I am basing this off of the notes below:
[image]

dayalannair, 2022-06-14:
You did mention this in a previous message which I think answers the question:
> It is the general procedure for test data, but in the real world you'll add sufficient gain to your receive chain to place the natural thermal noise at the correct level.
In this case I think the FFT is to be improved another way.

jpt13653903, 2022-06-14:
You mention that you're feeding it real data...  This bothers me, because your simulation output is quite clearly not symmetrical, meaning you've been feeding it complex data...

Unless you mean "real" as in "not emulated"?

jpt13653903, 2022-06-14:
Yes -- the gain in the receiver places the thermal noise level 16 dB above quantisation, which is plenty.  They'll get better dynamic range if they place it at Q+12 dB, but that's probably low on their priority list.

dayalannair, 2022-06-15:
Great. The notes above are from lecture slides, though now I think I should try to find this out for my device (of which they don't seem to want to give out too much info)

closed 2022-06-16 by dayalannair