dayalannair/FMCW_FFT_Radar #8: Python script

dayalannair, 2022-06-17:
Hi @jpt13653903 

I believe the FPGA is configured correctly. I am sending the FFT data over the UART (fire and forget) as 8 byte packets with the first half representing the Real portion of the sample and the second half = Im part. I am extracting the data in using a loop as shown below. 

1. Since the data is unsigned, I tried both int (i) and long (l) for unpacking the data which seems to work (need to find out if 2s complement, though it should be). Is this suitable?
2. After viewing the first 3 header bytes on each loop iteration, it seems that the data is not synced correctly. I added a condition/while loop to wait for the sync, though I feel the speed of transmission affects this - the loop does not detect the sync byte.
3. The FPGA executes the FFT and sends data on button push, and the process can be reset using the reset line. I have not simulated the reset cycle so if a reg was not reset, this may cause issue (though it does not seem like this is the issue). Is this fine? Pyserial blocks until data arrives so once the button is pushed, the FFT computation and data transfer is virtually instant.

LOOP in python script (permalink nice 🥇 ):
https://github.com/dayalannair/FMCW_FFT_Radar/blob/1554a1569fcff1e5774c2f9164f97e58a8421ac4/Python_scripts/FFT_read.py#L22-L45

jpt13653903, 2022-06-17:
Your Python looks ok -- I would look for the problem elsewhere...  Especially seeing as you're not getting the sync byte.

So I've opened the full Python file and looked elsewhere...

- You define the sync byte as a byte array, but your comparing it after casting to `str`...  I can't remember the exact Python comparison rules, but are you sure that it will ever return `True` when you actually receive the sync byte?  Does the type difference matter?  As a sanity check, have Python print out `first_byte` in the `while` loop.  A 0x55 byte is a 'U' character.
- My suggestion would be to dump what you actually receive to file and have a look.  You can either dump the binary received stuff directly and open with a hex editor, or you can get Python to write the values to file in hex (and use hex -- don't write the "characters" or "strings")

I would also format everything into larger packets (your header overhead is quite large in comparison with the payload size of 8 bytes), but that's an optimisation problem, and not what's wrong with your solution at this point.

On a side note (use it / don't use it...):

- Pull `str(struct.unpack_from('<c', s.read(1), offset=0)[0])` into a `getChar()` function (or similar), it'll make your code instantly more readable, and easier to optimise later, such as using `unpack` instead of `unpack_from`.  Indeed -- I don't think you need to `unpack` at all, simply read the character and cast the resulting byte to a string.
- Similarly, write a `getInt()` or similar function for the integers.  Even if you're busy writing test code that you intend to throw away and / or optimise later, having something that's easier to read, and making it such that fixing a bug in one place automatically fixes the bug everywhere else is super useful.  Also makes for less typing ;-)
- Use f-strings.  You can replace the above with `print(f'dest = {s.read(1)}\n')` or similar.  It's a bit cleaner than the "+" all over the place, and you can do really nifty formatting with it.
- I've also been stuck on the traditional `f = open(...)` ... `f.close()` pattern, but the more modern `with open(...) as f:` pattern is super useful and makes everything surprisingly nice and clean.  It will also move your "close" command (which is in the wrong place at the moment) to where it's supposed to be.

jpt13653903, 2022-06-17:
> Since the data is unsigned

Huh -- Why would FFT output be unsigned?

jpt13653903, 2022-06-17:
> though I feel the speed of transmission affects this

It shouldn't, unless you have long wires and you're getting reflections, which I doubt...  You can try dropping the Baud rate anyway, just to check

jpt13653903, 2022-06-17:
On your last point.  I haven't checked your code that does this, but just sanity-check that your reset cycle does not make your Python side confused.  It essentially has a buffer on the receive side, so you might be receiving old junk instead of the new data you want.

I don't really think that this is your problem, however.  If your "read" call is blocking before reading the first byte, then the queue is already empty.

dayalannair, 2022-06-18:
> 

Great advice! thank you. 

> > Since the data is unsigned
> 
> Huh -- Why would FFT output be unsigned?

Apologies, I meant to say that using '<I' in the unpack function casts to unsigned

dayalannair, 2022-06-18:
### Data extraction

1. Changed wrapper to output data starting at most significant BYTE: ensures 8 bytes are ordered correctly in the packet.
2. Need to ensure sign extended output is correctly converted into an integer - find python function or create one (requires output to be stored as binary or hex)

dayalannair, 2022-06-18:
2 could be a recursive function

jpt13653903, 2022-06-18:
> ### Data extraction
>
>     1. Changed wrapper to output data starting at most significant BYTE: ensures 8 bytes are ordered correctly in the packet.
>
>     2. Need to ensure sign extended output is [...]

The struct package does this for you.  You don't need to write your own.  You do need to make sure that you send the correct format though:

- Intel CPUs (including the Python struct package) reads little endian (i.e. LSB first)
- If you're sending 16-bit, tell the struct package to extract 16-bit.
- If you're extracting 32-bit, make sure that the firmware is sign-extending to 32-bit

dayalannair, 2022-06-18:
Will look into this, though seems straight forward to implement. The UART module we designed does send LSB first and the firmware is extending to 32 bit:

[image]

Im packagjng the data like this:

sync + Header + Im part (4 byte) + Re part(4 byte)
The only order I can change is the byte order: sending the LS bytes first (commented out below). Another option is to try bit reversed order of FFT output
https://github.com/dayalannair/FMCW_FFT_Radar/blob/2a3f49810281a7a5811f9fb744ebc877b2f93ae7/FMCW_FFT_Radar.srcs/sources_1/new/FIFO_stream_wrapper.v#L133-L134

dayalannair, 2022-06-18:
Could not find info on struct unpack for sign extended integers, will proceed with a custom function

dayalannair, 2022-06-18:
If I output unscaled, that means the data can be a maximum magnitude of 2^31? for a sign extended number, how do we know if the MSBs are 1111 or that is just the sign extension? I think there must be a minimum size. I think its 24 bits and anything more is sign extension.
[image]

dayalannair, 2022-06-18:
### Problem (roughly) solved

Changed little to big endian as shown in code below. Result is rough though.
[image]
https://github.com/dayalannair/FMCW_FFT_Radar/blob/fee87f23e05a830ab6a2b9bb5336c2e8a0a3fb44/Python_scripts/unpack_pkt.py#L4-L12

jpt13653903, 2022-06-18:
The `>>` operator in Verilog does a logical shift, not an arithmetic shift.  It fills the MSb side with zeros.  You can use `<<<` and `>>>` for arithmetic shift.

This said, it doesn't matter in this case -- you're just sending data, so the signedness doesn't matter at this point.

----

> Could not find info on struct unpack for sign extended integers, will proceed with a custom function

The struct package won't call it "sign extended integers".  It's simply "signed" and "unsigned".  According to the [documentation](https://docs.python.org/3/library/struct.html), you're looking for "<i".  "<I" is unsigned 4-byte.

----

> FINAL EDIT: The first one must be correct: based on the header with MS byte = sync, the first byte must be the MS byte.

The header has nothing to do with the payload.  It is independent from the payload.  Don't confuse them.  Send the data LSB first (note that the "B" is big, meaning "byte" -- little "b" means "bit") and decode signed little-endian (i.e. "<i").

You can, of course, leave it as big endian (as you have it at the moment), it just makes your Python-side work a bit harder, because you're using big endian on a little endian machine, and the decoder must do a translation.  It's generally a good idea to keep to the native endianness (if you have an option).

----

> If I output unscaled, that means the data can be a maximum magnitude of 2^31? ....

You might want to read up on how 2's compliment (and sign extension) works...  I recommend starting with [the Wikipedia article](https://en.wikipedia.org/wiki/Two%27s_complement).

----

> Result is rough though.

That is more than "rough".  It is plain wrong.  I'm sure you'll figure it out though.

dayalannair, 2022-06-19:
Okay I should be sorted. After needing to change struct to Big E I realised I need to swap the order on FPGA. Apologies for lack of clarity on the header thing, I meant I used it as a reference to find the endianess based on the known order of the bytes (as opposed to random -ish data).

- [ ] LSB first on FPGA
- [ ] read up on >> vs. >>>
- [ ] read up on sign extension and 2s compl

closed 2022-06-19 by dayalannair