dayalannair/FMCW_FFT_Radar #9: FIFO missed first FFT output sample

dayalannair, 2022-06-18:
Need to fix bug
[image]

dayalannair, 2022-06-18:
FIXED: bram write enable going high one clock late. used XOR gate with write_disable and FFT output valid line as the valid line goes high at the correct time but remains high (continously outputting FFT result). 

New:
First sample outputted for 2 clock cycles, could be BRAM latency. no problem as baud rate slower
[image]

jpt13653903, 2022-06-18:
> no problem as baud rate slower

Just a hint of advice: don't rely on these kind of hacks.  It will bite you in future.  BRAM latency is particularly nasty, because you need to keep track of when you make it wait and when not.  Best to hook up your output "Ready" to the BRAM clock enable port (not sure if you do that -- I haven't checked the code).  Gating the BRAM clock with the output ready makes sure that everything runs nice and deterministically even when your output randomly makes it wait.

dayalannair, 2022-06-19:
The IP seems to only have an enable port, no clken. Options are:

1. connect ready to port enable (for both read and write?) I see you say only the ready line is to be connected, not valid
2. AND ipClk and output ready and feed this to BRAM

jpt13653903, 2022-06-19:
When you run the wizard to get a BRAM block, there's a tick-box you need to tick to get access to the clock enable ports

dayalannair, 2022-06-19:
Yes I could not find that...usually all IP has it but I think it may be linked with the port enable. will check datasheet

jpt13653903, 2022-06-19:
I'm on Vivado 2017, but I'm sure the newer ones will be similar...  Mine generates the clock-enables by default.  They call it `ena` and `enb` though, not `clken` like Quartus does (which I assume you mean when you say "port enable"?).  These are unrelated to the `wea` and `web` signals, which are "write enable" lines for the two ports.

[image]

jpt13653903, 2022-06-19:
With Avalon interfaces you can simply hook up the ready to the enable and call it a day.  With AXI you have to be a bit more careful, because the handshake runs the other way round.

You generally have an if-statement gating your state machine so that you can handle the waiting correctly.  Hook up the body of that if-statement to the BRAM enable.

dayalannair, 2022-06-19:
> I'm on Vivado 2017, but I'm sure the newer ones will be similar... Mine generates the clock-enables by default. They call it `ena` and `enb` though, not `clken` like Quartus does (which I assume you [...]

Yes! I expected it to be aclken and bclken. Currently I just hold those high, will look into gating it. Seems low priority since the module is working correctly? Is it just for best practice? Cant determine from text message if you are empasizing or suggesting :grinning:

jpt13653903, 2022-06-20:
If it works, great.  Just don't do "AND ipClk and output ready and feed this to BRAM", that is bad.  Very bad.  If you do that then you're creating a new clock domain, with all the clock crossing related problems.

closed 2022-06-20 by dayalannair