dayalannair/FMCW_FFT_Radar #10: Comparator performance in Verilog

dayalannair, 2022-06-19:
Hi @jpt13653903 
I would like to know the difference in latency between a multi condition 'if' statement vs. a nested if statement. I believe a multi condition is generally faster as the comparison happens in parallel. If one condition takes a while to stabilise, both options above will be affected the same way.

dayalannair, 2022-06-19:
I will investigate

jpt13653903, 2022-06-19:
The latency is the same for both.  Both happen equally in parallel.

For the multiple conditions in a single if, all comparisons must be evaluated.  For the long if... else if... all conditions must be evaluated as well, because the final "else" is dependent on all previous conditions.

Draw the circuit, you'll see what I mean.

This is not true for a "case" statement, of course.  There you do not have priority encoded into the statement, so the compiler can make the assumption that only one case is true at a time.

jpt13653903, 2022-06-19:
Same is true for nested - the circuits are identical

dayalannair, 2022-06-19:
Indeed. I chose if here to avoid 1 clk latency when switching states and not sure if state machine is good here... will have a think

dayalannair, 2022-06-19:
FIFO will be converted to FSM

jpt13653903, 2022-06-19:
> ... prioritise read

I'm a bit confused on this one...  The read and write on a FIFO are generally independent...  You don't "prioritise" one above the other.

dayalannair, 2022-06-19:
Yes I think Im mistaken. Both if statements below happen in parallel, correct?
https://github.com/dayalannair/FMCW_FFT_Radar/blob/17e24c3bc241ca2a7bc5f9c610728672fcb2b7eb/FMCW_FFT_Radar.srcs/sources_1/new/FIFO_custom.v#L74-L89

jpt13653903, 2022-06-19:
Yes -- they're running in parallel.

You have one weirdness -- you're setting the `read_valid` in the write block.  This will cause you all sorts of nonsense.

dayalannair, 2022-06-19:
I have coupled the read and write blocks: read needs there to be atleast one sample in the buffer, hence it is gated by the write pointer. Ofcourse for a FIFO this is incorrect. Similarly, once atleast one sample is in the buffer (write state), read valid goes high to indicate data can be read. It works a little differently as the BRAM in//out ports are directly connected to the module output, so the module only controls addresses and write enable. The BRAM IP controls the rest (kind of symbiotic).

jpt13653903, 2022-06-20:
Still - you're assigning more than once to the same signal.

Move line 84 to 76 and put an else in front of the if

dayalannair, 2022-06-20:
Okay makes sense. And then line 77 must change from else if to just if (otherwise it wont be evaluated once wr addr>1), so all 3 things happen in parallel: read, write and read valid

jpt13653903, 2022-06-20:
Something along those lines...  Just don't assign to the same signal in more than one branch.

Even though Verilog (and circuit synthesis) does allow it, it results in really hard to find bugs.

In the words of Brown and Vranesic: always write code such that it is easy to visualise the resulting circuit.

closed 2022-06-20 by dayalannair