dayalannair/FMCW_FFT_Radar #11: WORKING!

dayalannair, 2022-06-19:
Had to correct issue where second data byte was being skipped. Correct testbench:
[image]
Result:
[image]
last little issue: last 2 samples are incorrect and had to be set to zero.
[image]
[image]

dayalannair, 2022-06-19:
Managed to get one less incorrect
[image]
Though FFT changes a bit and still looks better when the last 2 are set to 0. may resolve once the last value is fixed
 
[image]
TO DO:

- [ ] Re run sim which stores the FFT data and compare to the transferred data.

jpt13653903, 2022-06-19:
Much better :-)

The last two sample issue is probably timing-related (as in exact latencies of the various blocks involved)...

dayalannair, 2022-06-19:
Can I ignore the last sample or should I fix it? Also, I made the FIFO linear so it writes up to the last address then stops (does not circle back). Since the FIFO size is the same as the FFT point size this works fine. Can I leave it like that?

dayalannair, 2022-06-19:
It could have to do with read/write priority. Currently write has first priority, so the output valid line may go low late. I doubt this as both ifs should happen in parallel:
https://github.com/dayalannair/FMCW_FFT_Radar/blob/e974c4ca11427bb7dc9c61195f151ec2bfd38121/FMCW_FFT_Radar.srcs/sources_1/new/FIFO_custom.v#L75-L95
Simulating to the end takes a while since UART is slow (wouldnt mind tips i.e. if we can start sim at a later time point) and shows that the FIFO read ready stays high as the write byte count increases to one. Based on the code below, the valid line goes high but then no more data is placed on the line. Since the sim is slow, I will:

1. Change priority to read as write finishes early on
2. also add a counter to the multi condition if statement

[image]
https://github.com/dayalannair/FMCW_FFT_Radar/blob/e974c4ca11427bb7dc9c61195f151ec2bfd38121/FMCW_FFT_Radar.srcs/sources_1/new/FIFO_stream_wrapper.v#L110-L122

dayalannair, 2022-06-19:
Ive found that the last sample matches the first sample after doing some minor tweaks - this was not the case before. this just increases the center peak and smooths ripples:
### Without last sample:
[image]
With last sample where first and last samples --> RE = 411192, IM = 423290:
[image]
Question: should I leave the last sample as is, since it only affects transmitter feed through/center peak?

jpt13653903, 2022-06-19:
> I made the FIFO linear

I think I know what you mean, but if I understand you correctly you've technically implemented a buffer, not a FIFO queue (even though the behaviour is somewhat FIFO-like).

This is fine.

This said -- a FIFO queue for this use-case is generally a little larger than the FFT size, so that it can hold the start of the next sweep while your reading the previous.  A size of two sweeps are typical.

Also, why FIFO at the output of the FFT at all?  Why not simply hold it back with the "ready" line?

jpt13653903, 2022-06-19:
> Simulating to the end takes a while since UART is slow

Why on earth would you simulate with the UART in place?  You have much more patience than I do...

dayalannair, 2022-06-19:
> Ive found that the last sample matches the first sample after doing some minor tweaks - this was not the case before. this just increases the center peak and smooths ripples:
> ### Without last sample -> observe little peak on LHS of center peak:
> 
> [image]
> ### With last sample where first and last samples --> RE = 411192, IM = 423290:
> 
> [image] Question: should I leave the last sample as is, since it only affects transmitter feed through/center peak?

EDIT: the little peak is not a peak. the next value would be the last sample (fftshifted) which has been set to 0

dayalannair, 2022-06-19:
> > Simulating to the end takes a while since UART is slow
> 
> Why on earth would you simulate with the UART in place? You have much more patience than I do...

Was just checking the UART behaviour, yes I could exclude it

dayalannair, 2022-06-19:
> I think I know what you mean, but if I understand you correctly you've technically implemented a buffer, not a FIFO queue (even though the behaviour is somewhat FIFO-like).
>
> This is fine.
>
> This said [...]

I see. I can revert it to 512 depth though for this application one sweep should be fine? not sure if we need to bear in mind future work for this project. anyway should be quick to do

jpt13653903, 2022-06-19:
> Question: should I leave the last sample as is, since it only affects transmitter feed through/center peak?

Actually looks as if you have an off-by-one error on your address...  The left half of the FPGA version looks as if it's one sample to the left.

jpt13653903, 2022-06-19:
> I see. I can revert it to 512 depth though for this application one sweep should be fine? not sure if we need to bear in mind future work for this project. anyway should be quick to do - possibly trivial so I think I will leave it out :) but will mention in the report

ok

dayalannair, 2022-06-19:
> Actually looks as if you have an off-by-one error on your address... The left half of the FPGA version looks as if it's one sample to the left.

I think this is from setting the last value to 0...though now Im thinking that since the last value = the first value, the address may be wrapping back to index 0. The second figure, second last plot shows that when the last sample is added, it exceeds the expected peak magnitude. Strangely, the first sample does not do this when the last is set to 0, despite being the same value.

closed 2022-06-20 by dayalannair