dayalannair/FMCW_FFT_Radar #14: Last sample = First sample = Large number

dayalannair, 2022-06-19:
> Actually looks as if you have an off-by-one error on your address... The left half of the FPGA version looks as if it's one sample to the left.

I think this is from setting the last value to 0...though now Im thinking that since the last value = the first value, the address may be wrapping back to index 0. The second figure, second last plot shows that when the last sample is added, it exceeds the expected peak magnitude. Strangely, the first sample does not do this when the last is set to 0, despite being the same value.

_Originally posted by @dayalannair in https://github.com/dayalannair/FMCW_FFT_Radar/issues/11#issuecomment-1159788508_

jpt13653903, 2022-06-20:
Not what I meant with "one sample to the left".  Look at the figure again.  Even better, plot the FPGA results on the same figure as the Matlab results (with appropriate scaling to get them the same size).  You'll see that your left half is shifted.

dayalannair, 2022-06-23:
Will get on this... must be the fft outputting/buffering from address 1 instead of 0 or something
[image]

closed 2022-06-24 by dayalannair