dayalannair/FMCW_FFT_Radar #15: FIFO vs. Buffer

dayalannair, 2022-06-20:
FIFO - first data in is the first out. Type of buffer.
Buffer - holds data before use

I think the difference is that the FIFO wraps back to the first position once that data has been read, though this sounds like a circular buffer. The difference being that the data will not be overwritten once FIFO is full? A normal buffer fills and stops receiving while data is being sent out. 

FIFO is for async clocks, so data can be streamed continuously whereas a buffer fills, then has to empty and fill again. For this application it should be fine though a FIFO is preferable.

jpt13653903, 2022-06-20:
A circular buffer is one way by which you can implement a FIFO.  By itself it is not a FIFO, because you can use it for other stuff as well.  Sort of like a rectangle not being a square.

The reason I don't want to call your buffer a FIFO is because you're not using it as a FIFO (even thought it technically is one).  Your implementation doesn't really have the concept of "queue length".  You fill up the buffer, and then you read till the top, and then you start writing from the start again.  A FIFO (in my mind) is something where you can read and write completely independently, as long as the queue length is somewhere between 0 and full.

The async clock thing is only one use case (of many) of a FIFO.

dayalannair, 2022-06-20:
Makes sense. Mine technically reads and writes at the same time as soon as theres one element in the buffer, though it does not consider queue length