dayalannair/FMCW_FFT_Radar #18: FFT input stream needing too many zero pads

dayalannair, 2022-06-21:
[image]

jpt13653903, 2022-06-21:
Are you sure you "need" them, and that it's not simply the FFT IP that allows you to start inputting the next FFT's samples while the current one is busy?

dayalannair, 2022-06-21:
Im pretty sure if the FFT slave side remains ready, the master side will not output anything. Will confirm this sometime. Got another issue that ma be related: the master is only outputting 35 valid FFT outputs.
[image]

dayalannair, 2022-06-21:
If I cant find a fix, I will just add input and output BRAM buffers

dayalannair, 2022-06-21:
Theory: the testbench data bytes are just looping from 0 to 255 so maybe the faulty data/unusual is affecting fft length......unlikely though. Could also be the padding issue

dayalannair, 2022-06-21:
Guess I gotta rtfm. Found this , though need to ensure the context is correct. Seems like a buffer is the safest option as the FFT input does not seem to work in the same way.
[image]

dayalannair, 2022-06-21:
Toggling the valid line violates the timing reqs, so a buffer is needed to prevent this.
[image]

dayalannair, 2022-06-21:
### With Buffers
still needing an extra 256+131 ~ 400 samples

[image]
Need to check output/result to see if this is a problem