dayalannair/FMCW_FFT_Radar #22: Code submission

dayalannair, 2022-06-21:
Can I submit a link to this repo? Not too sure which files can be stripped off for vivado to run correctly (last time I tried that it broke). Or should I just submit the source code (.v and .sv)? this wont include the fft IP. And to clarify, there was ofcourse no expectation to design a custom FFT ?😄

jpt13653903, 2022-06-21:
No expectation of a custom FFT ;-)  Just getting their IP to play nicely is more than sufficient.

And yes - a link to the repo is good

jpt13653903, 2022-06-22:
Found it:

```
**/*.sim/*
**/*.runs/*
**/*.cache/*
**/webtalk*
```

This is of course in addition to my usual "root" `.gitignore`:

```
.vs
.vscode
*.swp
*.bak
~*
```

dayalannair, 2022-07-01:
I will add this and test it out some time 👍

dayalannair, 2022-07-09:
Changed the ignore to the one above and cloned to a new directory. Project size is still large. Seems like the gen folder is the largest.