dayalannair/FMCW_FFT_Radar #29: Output right shifted and incorrect/missing samples around 0 Hz

dayalannair, 2022-06-24:
For now, this bug is left for future work for two reasons:

1. It can be easily corrected/accounted for 
2. The bug is small yet not obvious
3. Time saving

Current focus is to send/receive multiple sweeps/FFTs which is just an outer loop in the python script

dayalannair, 2022-06-24:
### Circshift -1 units and set last 2 samples = 3rd last sample (hold)
[image]

jpt13653903, 2022-06-24:
So you actually have only one wrong sample -- not bad.

You still need to change your scale to log-scale...

dayalannair, 2022-06-24:
Yes, though here it was two. Managed to get it to be only the first sample thats wrong - which is actually just a duplicate of the second sample... I may be able to solve this as it means that the first sample in the buffer is read twice.

dayalannair, 2022-06-24:
### FIXED

closed 2022-06-26 by dayalannair