dayalannair/FMCW_FFT_Radar #31: MULTI SWEEP (FINAL DESIGN) WORKING

dayalannair, 2022-06-24:
1. convert set of sweeps to hex with padding to 16 bits and combined IQ data
2. Have FFT reset during sending of data
3. Feed in sweeps and store output as rows in textfile
4. plot against matlab

NOTE: FPGA version much slower due to UART baud rate. Future work: send all sweeps via faster comms (directly from radar) and store in buffer. Then use multi channel FFT and send to PC over fast comms. Can also process sweeps in real time.

See plot below. can extract any sweep for comparison. Target is a trolley moving towards then reversing away from radar
[image]

jpt13653903, 2022-06-24:
Yes...  We generally use multiple channels of 10 Gb Ethernet.

For the purposes of this project, however, the slow UART is fine.  You slowly load data into a staging buffer, then play it through the processing chain (i.e. your FFT) at the correct rate, storing the output into a second staging buffer.  You can then read out the result at leisure.

dayalannair, 2022-06-24:
Great! suspected Ethernet is best. Will mention as future work.

closed 2022-06-26 by dayalannair