dayalannair/FMCW_FFT_Radar #33: FPGA unconnected pins

dayalannair, 2022-06-25:
VIvado claims unconnected pins are a critical warning. I think you said before that lattice ties these to ground, is this the same in the vivado compiler? Would like to know if there are consequences, though the pins that I havent connected will have no effect on the system even is they are floating

jpt13653903, 2022-06-25:
Most FPGA compilers make undefined pins weak high...  Note: "undefined", not "unassigned".

In Vivado you have 2 options - either don't define them in the first place, or define them as "virtual" pins.

The fact that it throws an error is VERY good.  Quartus assigns a random pin, which could be used by some other hardware, and then drives it to whatever...

dayalannair, 2022-06-26:
Ahh okay I will leave them undefined. Thanks. For interest, what exactly is a virtual pin?

jpt13653903, 2022-06-26:
[Virtual Pins](https://support.xilinx.com/s/article/58609?language=en_US)

closed 2022-06-27 by dayalannair