dayalannair/FMCW_FFT_Radar #34: Good verilog listing style

dayalannair, 2022-06-25:
Hi @jpt13653903 
I would like to know if you've come across such a thing and if so, could you please share it here. Thanks

dayalannair, 2022-06-25:
I have one that works, though it throws a compiler error when  [] is used instead of {} here:

`\begin{lstlisting}[style={verilog-style}, label = {lst:rcv_state}, caption = {Receiving raw data}]`

though this doesnt seem to affect the result:
[image]
Error rel to square brackets:
[image]

If I use {}. no formatting occurs:
[image]

jpt13653903, 2022-06-25:
Don't use Verilog "style".  Use Verilog "language".

For examples, have a look at [my template](https://github.com/jpt13653903/LaTeX_UCT_Report)

dayalannair, 2022-06-26:
I am defining the style using the Verilog language--> this should work?
[image]

dayalannair, 2022-06-26:
The issue is with curly and square brackets - latex is expecting some control/command within them. Tried escaping which didnt work. just printed the \ in the listing. The listing output is fine, maybe it generates an error then prints as normal

jpt13653903, 2022-06-26:
I've cloned your repo in the hope that I can reproduce your problem...  No LaTeX code...?

I think you're trying to over-define the language, which makes it confused.

Really -- have a look at my [UCT Report Template](https://github.com/jpt13653903/LaTeX_UCT_Report/blob/master/Template/Listings.tex).

- Line 2 loads all the languages that I intend to use
- Lines 24 to 30 sets the style (globally, not per language)
- Then follows a section of helper-macros
- Line 183 to 202 defines the `Verilog` and `Verilog_float` environments.

I don't seem to have a Verilog example in there, but you can see how to use OpenCL (with both curly and square brackets) [here](https://github.com/jpt13653903/LaTeX_UCT_Report/blob/cf22cdd8520ec5a17ce4590ced827b814fbcf5ef/Report/Chapters/Design.tex#L72).

dayalannair, 2022-06-26:
I am using Overleaf, which wants me to use a 7 day trial to link with git :(. Ive added a zip file of the current state of the doc here:
https://github.com/dayalannair/FMCW_FFT_Radar/blob/master/EEE5118Z_Report_NRXDAY001(1).zip

I wouldnt mind using your code, though the error message doesnt seem to have any consequences. I'd also like to solve the issue for future eventualities/out of interest instead of avoiding it. I will give your code a try, though, as it does seem very straight forward.

jpt13653903, 2022-06-26:
Ah - Overleaf...

I personally prefer using a native compiler (MikTeX on Windows). Also, although I've never used it myself, I saw some really odd behaviour when looking at what other people are doing.

I'll have a look at your zipped version when I'm back at my PC

jpt13653903, 2022-06-26:
Right -- Couple things:

- You are importing the same package multiple times -- don't do that: it causes namespace-related clashes and causes all sorts of nonsense.
- You're using both `listings` and `minted`.  It's generally a bad idea to use two packages that do almost the same thing -- they tend to clash and cause problems.  In addition -- it's generally bad style to have two different styles in the same report, so if you do use both `listings` and `minted`, make sure that their output styles are the same.
- I managed to reproduce your problem.  I then changed the style name from `verliog` to `my_verilog` and the problem went away.  I think you had a name clash between the language and the style.
- Be a bit more liberal with your empty lines -- it makes the code easier to read, especially around section headings and figures and such.

The attached one works:  [bare_jrnl.zip](https://github.com/dayalannair/FMCW_FFT_Radar/files/8987104/bare_jrnl.zip).  I also corrected a bunch of other stuff -- you can run a diff to see what I've done.

dayalannair, 2022-06-26:
Great! Thanks. The name was originally 'verilog-style' and it still had the issue so maybe something else fixed it? 

Yes. Minted looks better (atleast that py  one) and I thought it may be nice to show different languages differently, though I think Ill use lstlistings (or whatever you've changed it to if you did).

jpt13653903, 2022-06-27:
Maybe it doesn't like hyphens in names, but that is pure speculation...

If you like the `minted` look, why not use it for everything?  Why use `listings` at all?  I like `listings` because it doesn't come with a Python dependency.

And if you want to use `listings`, you can style it to look exactly like `minted`...  Just saying...

My point is: don't import libraries just for the sake of importing libraries, especially in LaTeX.  It does not provide the namespace isolation that most programming languages provide, so you often get clashes.  Pulling in more libraries also result in more dependencies, which is not always the greatest idea...

And to make matters even worse, the order in which you import libraries is important.  Generally, import the `hyperref` last, for example.

dayalannair, 2022-06-27:
I found that the error is specifically this line in the style 'object':
` %moredelim=*[s][\colorIndex]{[}{]},`

closed 2022-06-28 by dayalannair